audeme/MOVIArduinoAPI #4: Use callsign as generic keyword spotter

fractor, 2016-01-19:
User requested a way to make MOVI only listen to the callsign and not prompt for a sentence afterward.
Possible implementations:
- Add ABORT command to the low level interface (firmware patch). ABORT would quit listening to a sentence.
- Add a CALLSIGNONLY <ON/OFF> command to the low level interface (firmware patch). So MOVI knows not to listen to a sentence after detecting a call sign.

oziee, 2016-06-11:
i would like to see CALLSIGNONLY too.. i have now added a MOVI to another project and found that i just want to say MOVI and have it send a response to the arduino that the callsign was called

fractor, 2016-06-11:
You can use the above script but MOVI will still try to listen for a sentence after the CALLSIGN is detected. That's why the 1.1 firmware update will come with an ABORT command to abort that early. Your script will then look like this:

void loop() // run over and over
{
signed int event=recognizer.poll(); // Get result from MOVI, 0 denotes nothing happened, negative values denote events (see docs)
if (event==BEGIN_LISTEN) { // This is returned at the start of listening
digitalWrite(led, HIGH); // LED on.

recognizer.say("Yes"); // Speak a response its listening
}
if (event==END_LISTEN) { // This is returned at the end of listening
digitalWrite(led, LOW); // LED off 
recognizer.abort(); // Force going back into callsign listening mode.
}
}

oziee, 2016-06-12:
would the abort() call be in the BEGIN_LISTEN test.. calling abort would then call END_LISTEN ?

Ill be testing 1.1 when it comes out:)

fractor, 2016-06-12:
You are totally right. I got the events confused. And yes, it'll through END_LISTEN. This is how it would look like:

void loop() // run over and over
{
signed int event=recognizer.poll(); // Get result from MOVI, 0 denotes nothing happened, negative values denote events (see docs)
if (event==BEGIN_LISTEN) { // This is returned at the start of listening
digitalWrite(led, HIGH); // LED on.
recognizer.abort(); // Force going back into callsign listening mode.
recognizer.say("Yes"); // Speak a response its listening
}
if (event==END_LISTEN) { // This is returned at the end of listening
// this will be triggered immediately on the abort though. So not sure if you want to keep it.
digitalWrite(led, LOW); // LED off 

}
}

In anyways, waiting for 1.1 is probably best.

fractor, 2016-07-20:
@ecm-bitflipper wrote:
My dream feature is for my MP3 player to announce a special greeting at CALLSIGN_DETECTED but before MOVI listens for a sentence. I tried yet another approach where I did a finish() after BEGIN_LISTEN, waited for the bogus response to come back from MOVI, then did ask() after playing my MP3.  Not only was this unacceptably sluggish, but MOVI was still hearing a fragment of my callsign acknowledgement MP3, usually just the tail end of it when I listened in micdebug mode, but enough to confuse MOVI and not allow for accurate sentence matching.  I couldn't figure out the timing issue - it seemed very strange.  Regardless, the whole exercise takes too long for the interaction with MOVI to be smooth.

fractor, 2016-07-26:
Added WordSpotter2 example.

closed 2016-07-26 by fractor