audeme/MOVIArduinoAPI #18: MOVI announces boot-up message at power-up even with welcomeMessage(false)

ecm-bitflipper, 2016-10-04:
The function to silence the MOVI boot-up welcome message does not seem to work.  I tried the following code snippit and can tell that something inside init() is causing the boot-up welcome message to announce because I can see "Starting MOVI" on my LCD the entire time MOVI is speaking the welcome message, and I don't see "Starting MOVI2" until after the announcement is made.  I tried both welcomeMessage(false) and SYSTEMMESSAGES OFF but neither silence the boot-up "MOVI is booting...  MOVI is ready, the callsign is ...." announcement.

`  lcd.setCursor(0,1);`
`  lcd.print("Starting MOVI   ");`

`  movi.welcomeMessage(false);             // silence MOVI's welcome message`
`  movi.sendCommand(F("SYSTEMMESSAGES"), F("OFF"));`
` `
`  movi.init();`

`  lcd.setCursor(0,1);`
`  lcd.print("Starting MOVI2  ");`

fractor, 2016-10-05:
init() resets the welcome message setting.
The best thing is to do is to set the message flags AFTER the init() method.

ecm-bitflipper, 2016-10-05:
So does that mean that by design, I will always hear the "MOVI is booting... etc." at first power-up of the Arduino/MOVI and that is by design not possible to override?

fractor, 2016-10-05:
The boot message cannot be overriden. That's by design. A solution around this would be to delay turning on the speaker that's connected to MOVI. Another way is to hack the SDcard...

ecm-bitflipper, 2016-10-05:
I'd much rather tweak the software (SDCard) than build more hardware gadgets.  Can you give me a hint as to what would need to be "hacked"?

fractor, 2016-10-05:
This is on your own though.
  
Mount the MOVI Root partition (with Linux) and uncomment line 24 in etc/init.d/hostname.sh, i.e. change: 
/usr/bin/espeak -vdefault+m3 -p 40 -s 160 "mohvee booting" &
to
# /usr/bin/espeak -vdefault+m3 -p 40 -s 160 "mohvee booting" &

ecm-bitflipper, 2016-10-05:
Thanks, I'll try it next time I pull my Arduino out of its hiding place.  No worries, I won't damage anything, I'm very comfortable editing shell scripts in Linux.

closed 2017-02-04 by fractor

ecm-bitflipper, 2018-12-30:
Hard to believe it's been more than 2 years since I posted this.  Anyway, I finally had another reason to pull out my Arduino so I tackled this as well.  I imagine you were trying to tell me to comment out that line but github ate the hash-tag.  But then as I was looking at it, I figured this would only eliminate the boot announcement, but not the "MOVI is ready, the callsign is..." part.  Since I don't use MOVI's speak feature at all (all my voices are done with a separate MP3 board), I got brave and renamed /usr/bin/espeak to /use/bin/espeak.backup, hoping that wouldn't cause the bootup to crash.  Luckily all is well, MOVI is quiet during its boot phase and becomes ready without any errors.

Just thought I'd let you know, in case you were curious.

audeme, 2018-12-30:
All messages other than the boot message can be quieted by these two API methods: 
    // Turns the spoken responses as a result of recognition events (e.g. silence or noise) on or off.
    void responses(bool on);
    
    // Turns off the spoken welcome message indicating the call sign.
    void welcomeMessage(bool on);

Just the boot message needs the hack we talked about. However, whatever works for you...

ecm-bitflipper, 2018-12-31:
If you go back to the beginning of the thread, you'll see that at first power-up, the init resets the welcome message ("MOVI is ready the callsign is..."), and so it will be heard at first startup no matter what you do with welcomeMessage.  That's why I renamed espeak.  I wouldn't recommend it necessarily, only for unique applications where one wants to hear the BEEP-BEEP after the callsign is recognized, but absolutely no speaking.