audeme/MOVIArduinoAPI #27: NRF24L01 + MOVI

drolsen, 2019-11-05:
There appears to be some kind of Serial conflict with using RF24 library and Movi.
I know it is a lot to ask to look into, but not being able to use Movi with an arduino radio transceiver project, is very limiting disheartening to a lot of builders (me included).

In the setup, soon as Ears.init() is called (before or after any other code) my RF24 fails to send / recive over it's RX/TX. Without Ears.init(), my RF24 comes back to life.

```c++
#include <MOVIShield.h>
#include <RF24.h>

MOVI Ears(true);
signed int Perceived;

/* Wireless communication to remote control */
RF24 Radio(3, 2); // CE, CSN
const byte address[6] = "R2D2";
unsigned long RadioWait = 0;

// Max size of this struct is 32 bytes - NRF24L01 buffer limit
struct Data_Package {
  byte button1;
  byte button2;
  byte button3;
  byte button4;
};
Data_Package data; // Create a variable with the above structure

void setup()
{
  Serial.begin(9600);
  Ears.init(); // This line seems to kill NRF24L07 from using RX/TX

  /* Wireless Setup */
  Radio.begin();
  Radio.openReadingPipe(0, address);
  Radio.setPALevel(RF24_PA_MIN);
  Radio.startListening();
  WirelessWait = millis();
}

void loop()
{
  RadioPing(); 
}

void RadioPing() {
  while (millis() > RadioWait + 100) {
    if (radio.available()) {
      Serial.println("Ping!");
      radio.read(&data, sizeof(Data_Package));
    }
    WirelessWait = millis();
  }
}`

I'm really hoping to get this resolved cause I'm planning on featuring MOVI as the center peice to a shield I'm releasing to astromech.net R2D2 builder group with lots of people in great anticipation.

Thank you!!

drolsen, 2019-11-08:
This is a non issue after all, I was able to locate the LowLevelInterface example which bypasses SoftwareSerial for hardware serial / jumper pins to my Arduino Mega serial3 pins (Serial1 or Serial2 would also work). With this in place I no longer have a 3 second lag on my NRF24L07.

Thanks!

closed 2019-11-08 by drolsen

yepher, 2021-11-19:
@drolsen I am just getting ready to use MOVI with an NRF24L07 (waiting on all the parts to arrive). Do you have any tips or sample code for getting up and running quickly with Arduino, Movi and NRF24L07?