audeme/MOVIArduinoAPI #28: sending diacritic signs to the espeak

pawkon, 2020-08-19:
Because there is no answer on movi forum then I'm writing here as a bug? Movi is able to use espeak in different languages but is there any way of sending diacritic signs to the espeak to pronounce them right? For example in german like : ß or in Russian: ц or in Polish ź etc? Now sending them from arduino code they are pronounced like \u283 etc.

fractor, 2020-08-19:
Unfortunately, Arduino only handles 7bit ASCII.

pawkon, 2020-08-20:
I'm sorry but I don't understand ... Arduino Serial handles also special chars like in this simple code

`void setup() {`
`  Serial.begin(115200);`
`}`
`void loop() {`
`  Serial.println("Cześć");`
`  Serial.println("\u0627\u0644\u0639\u0631\u0628\u064A\u0629");`
`  delay(3000);`
`}`

so when running putty on arduino serial it is showing exactly what the code is sending. So the problem in my opinion is on the movi side

fractor, 2020-08-31:
Yes but they don't reach our board....

Gerald

<http://www.gerald-friedland.org>
<http://www.gerald-friedland.org>
<http://www.gerald-friedland.org>
<http://www.gerald-friedland.org>
<http://www.gerald-friedland.org>
<http://www.gerald-friedland.org>
<http://www.gerald-friedland.org>
<http://www.gerald-friedland.org>
-- <http://www.gerald-friedland.org>
Dr. Gerald Friedland <http://www.gerald-friedland.org>
Founder and CTO, Audeme LLC <http://www.gerald-friedland.org>
http://www.audeme.com <http://www.gerald-friedland.org>

pawkon, 2020-09-01:
Is there any workaround ?